jsk-ros-pkg/jsk_model_tools #108: jaxon.lが上手く生成されない

Tnoriaki, 2015-04-23:
catkin build を行なってjaxon.lが上手く生成されず
roseus jaxon-interface.l
を行なって
(jaxon-init) 
を行うと以下のようなエラーメッセージが出ます。
/opt/ros/hydro/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: float vector expected in (send-message self (class . super) :init-ending)

/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models
にあるjaxon.lを見ると
(defmethod JAXON-robot )の　(:init-ending) の部分が生成されていません。(4968行目以降がない)

どうすればいいでしょうか？
euscolladaのバージョンは
Version: 0.1.11-0precise-20150409-0928-+0000
でrosはshadow-fixedのaptレポジトリからバイナリをインストールしてきました

k-okada, 2015-04-23:
以下の文章が日本語にみえたら試してみて下さい．見え無かったら，もう１，２日待ってみて下さい．

http://repos.ros.org/repos/building/pool/main/r/ros-hydro-openhrp3/ros-hydro-openhrp3_3.1.8-0precise-20150422-1213-+0000_amd64.deb
を無理やりdpkg -i して，さらにrtmros_commonに以下のパッチをあてててどうなるかおしえてください．
https://github.com/start-jsk/rtmros_common/pull/708

Tnoriaki, 2015-04-23:
dpkg -i ros-hydro-openhrp3_3.1.8-0precise-20150422-1213-+0000_amd64.deb
をして
hrpsys_ros_bridge/cmakeに変更を加えた後に、
hrpsys_ros_bridge_tutorials/　で　
　git clean -xfd
　rm ~/ros/hydro/build/hrpsys_ros_bridge_tutorials -rf
　catkin build --this
を行いましたがビルドは通らず途中までで出来たjaxon.lも
変更を加える前の時と変わりませんでした

k-okada, 2015-04-23:
ログを見せて下さい．

Tnoriaki, 2015-04-23:
rosbuild2回やっています
```
takasugi:hrpsys_ros_bridge_tutorials(master)$ catkin build --this
-------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/hydro
Workspace:                   /home/takasugi/ros/hydro
Source Space:       [exists] /home/takasugi/ros/hydro/src
Build Space:        [exists] /home/takasugi/ros/hydro/build
Devel Space:        [exists] /home/takasugi/ros/hydro/devel
Install Space:     [missing] /home/takasugi/ros/hydro/install
DESTDIR:                     None
-------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
-------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
-------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------- 
Found '30' packages in 0.1 seconds. 
Starting ==> hrp2_models                                                                                                                                                
Finished <== hrp2_models                 [ 0.5 seconds ]                                                                                                                
Starting ==> hrpsys_ros_bridge_tutorials                                                                                                                                
                                                                                                                                                                        
[hrpsys_ros_bridge_tutorials] ==> '/home/takasugi/ros/hydro/build/hrpsys_ros_bridge_tutorials/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/takasugi/ros/hydro/build/hrpsys_ros_bridge_tutorials'
Scanning dependencies of target HRP2JSKNTS_model_generate
Scanning dependencies of target HRP2JSK_model_generate
Scanning dependencies of target HRP2JSKNT_WH_SENSORS.urdf_generate
Scanning dependencies of target HRP2JSKNTS_xacro_model_generate
[  1%] Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP3HAND_R.dae
[  2%] [  3%] [  4%] Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNTS.dae
Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP3HAND_R.dae
Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSK.dae
The "equivalentInertia" field of the Joint node is obsolete.
The "equivalentInertia" field of the Joint node is obsolete.
Humanoid node
Humanoid node
Joint node WAIST
Joint node WAIST
  Segment node RARM_LINK6
  Segment node RARM_LINK6
  Joint node R_THUMBCM_Y
      Segment node R_THUMBCM_Y_LINK
  Joint node R_THUMBCM_Y
      Joint node R_THUMBCM_P
      Segment node R_THUMBCM_Y_LINK
          Segment node R_THUMBCM_P_LINK
      Joint node R_THUMBCM_P
  Joint node R_INDEXMP_R
          Segment node R_THUMBCM_P_LINK
      Segment node R_INDEXMP_R_LINK
  Joint node R_INDEXMP_R
      Joint node R_INDEXMP_P
      Segment node R_INDEXMP_R_LINK
          Segment node R_INDEXMP_P_LINK
      Joint node R_INDEXMP_P
          Joint node R_INDEXPIP_R
              Segment node R_INDEXPIP_R_LINK
          Segment node R_INDEXMP_P_LINK
          Joint node R_INDEXPIP_R
  Joint node R_MIDDLEPIP_R
              Segment node R_INDEXPIP_R_LINK
[... 3532 lines cut ...]
  Joint node RLEG_JOINT0
      Segment node RLEG_LINK0
      Joint node RLEG_JOINT1
          Segment node RLEG_LINK1
          Joint node RLEG_JOINT2
              Segment node RLEG_LINK2
              Joint node RLEG_JOINT3
                  Segment node RLEG_LINK3
                  Joint node RLEG_JOINT4
                      Segment node RLEG_LINK4
                      Joint node RLEG_JOINT5
                          Segment node RLEG_LINK5
                            ForceSensorrfsensor
  Joint node LLEG_JOINT0
      Segment node LLEG_LINK0
      Joint node LLEG_JOINT1
          Segment node LLEG_LINK1
          Joint node LLEG_JOINT2
              Segment node LLEG_LINK2
              Joint node LLEG_JOINT3
                  Segment node LLEG_LINK3
                  Joint node LLEG_JOINT4
                      Segment node LLEG_LINK4
                      Joint node LLEG_JOINT5
                          Segment node LLEG_L

k-okada, 2015-04-23:
https://github.com/euslisp/jskeus/pull/200/files
みたいに，nilでboundingboxをつくろうとしているももの対応が必要な気がします．
とありあえずこれをコピーして/opt/ros/hydro/share/euslisp/jskeus/irteusみたいなところにいれて，
https://github.com/start-jsk/rtmros_tutorials/pull/254
にあるように，irtrobot.l をloadすればうまく動くと思う．

あと，https://github.com/start-jsk/rtmros_tutorials/pull/244を使っています．

k-okada, 2015-04-23:
https://github.com/euslisp/jskeus/pull/200/files
は
https://github.com/euslisp/jskeus/pull/189
で対応ずみかな．さっきみたいにdebをもてきてもいいし，1.0.5のirtrobotをoptにいれてloadしてもいいかな．
--
◉ Kei Okada

2015-04-23 19:01 GMT+09:00 Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>:
> https://github.com/euslisp/jskeus/pull/200/files
> みたいに，nilでboundingboxをつくろうとしているももの対応が必要な気がします．
> とありあえずこれをコピーして/opt/ros/hydro/share/euslisp/jskeus/irteusみたいなところにいれて， [...]

k-okada, 2015-04-23:
あとは，おまじないで，https://github.com/jsk-ros-pkg/jsk_model_tools/pull/109
しましたが，関係無かったかな．
--
◉ Kei Okada

2015-04-23 19:10 GMT+09:00 Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>:
> https://github.com/euslisp/jskeus/pull/200/files
> は
> https://github.com/euslisp/jskeus/pull/189
> で対応ずみかな．さっきみたいにdebをもてきてもいいし，1.0.5のirtrobotをoptにいれてloadしてもいいかな．
> --
> ◉ Kei Okada
>
>
>
> 2015-04-23 19:01 [...]

snozawa, 2015-04-23:
init-endingのmake-bounding-boxでエラーがでるものは
https://github.com/euslisp/jskeus/pull/189
で対応済みです．

ただし，このエラーは足先end-coords高さが足裏リンク形状の点の高さより大幅にズレてる場合におきるようなので，
`cd ~/prog/euslib/rbrain/jaxon && svn up`
と
`roscd hrpsys_ros_bridge_tutorials && git pull`
が足りてないのかもしれません．
（今手元のおそらく最新jaxonでは，end-coordsと足形状があってるっぽかったです）

Tnoriaki, 2015-04-23:
irtrobot.l をwgetしてそのirtrobot.lをloadすることでroseus jaxon-interface.lの後
(jaxon-int)
をすることが出来るようになりました。

ただ、その後
rtmlaunch hrpsys_ros_bridge_tutorials jaxon.launch
を行うと
```
Invalid <arg> tag: $(find pkg) command only accepts one argument [find ]. 

Arg xml is <arg default="$(find )/models/JAXON.xml" name="PROJECT_FILE" unless="$(arg NOSIM)"/>
```
というエラーメッセージをはいてシミュレーターが起動しなくなりました
hrpsys_ros_bridge_tutorials/launch以下のlaunchファイルで
(find )
のようにfind 以下がすべて空欄になっているみたいです。
どうすればいいでしょうか？

k-okada, 2015-04-23:
ちょっとなおしました．
https://github.com/start-jsk/rtmros_common/pull/708

Tnoriaki, 2015-04-23:
応急処置としてlaunchファイルにfind以下を書き込んだら
シミュレーターは起動しました。

k-okada, 2015-04-23:
> ちょっとなおしました．
> https://github.com/start-jsk/rtmros_common/pull/708

はなおらなかったでしょうか．

Tnoriaki, 2015-04-23:
ビルドはエラーがでて通りませんでした。
何回かビルドを繰り返す事でjaxon.lが生成できました

.launchはfind以下にプロジエクト名が代入されて正しく直っていました

k-okada, 2015-04-23:
2015-04-24 1:22 GMT+09:00 Takasugi Noriaki <notifications@github.com>:

> ビルドはエラーがでて通りませんでした。
> 何回かビルドを繰り返す事でjaxon.lが生成できました
>

エラーを貼ってください．

--
◉ Kei Okada

Tnoriaki, 2015-04-24:
今日やり直したら50分経ってもビルドが終わりませんでした
takasugi:hrpsys_ros_bridge_tutorials(master●)$ catkin build --this
-------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/hydro
Workspace:                   /home/takasugi/ros/hydro
Source Space:       [exists] /home/takasugi/ros/hydro/src
Build Space:        [exists] /home/takasugi/ros/hydro/build
Devel Space:        [exists] /home/takasugi/ros/hydro/devel
Install Space:     [missing] /home/takasugi/ros/hydro/install
DESTDIR:                     None
-------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
-------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
-------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------- 
Found '30' packages in 1.1 seconds. 
Starting ==> hrp2_models                                                        
Finished <== hrp2_models                 [ 0.3 seconds ]                        
Starting ==> hrpsys_ros_bridge_tutorials                                        
^C[build] User interrupted, stopping.                                                                                                                                     
                                                                                                                                                                        
[hrpsys_ros_bridge_tutorials] ==> '/home/takasugi/ros/hydro/build/hrpsys_ros_bridge_tutorials/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/takasugi/ros/hydro/build/hrpsys_ros_bridge_tutorials'
Scanning dependencies of target HRP2JSK_WH_SENSORS.urdf_generate
Scanning dependencies of target HRP2JSKNTS_xacro_model_generate
Scanning dependencies of target HRP2JSKNT_WH_SENSORS.urdf_generate
Scanning dependencies of target HRP2JSK_model_generate
[  0%] [  0%] Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.dae
[  0%] [  0%] Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.dae
Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.dae
Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.dae
Humanoid node
Humanoid node
Joint node BASE
Joint node BASE
  Segment node BASE_LINK
  Joint node J1
    Segment node J1_LINK
    Joint node J2
      Segment node J2_LINK
      Joint node J3
        Segment node J3_LINK
        Joint node J4
          Segment node J4_LINK
          Joint node J5
            Segment node J5_LINK
            Joint node J6
              Segment node J6_LINK
              Joint node J7
                Segment node J7_LINK
                Joint node HAND_L
                  Segment node HAND_L_LINK
                Joint node HAND_R
                  Segment node HAND_R_LINK
Humanoid node
Humanoid node
  Segment node BASE_LINK
Joint node BASE
  Joint node J1
  Segment node BASE_LINK
[... 3242 lines cut ...]
log file: /home/takasugi/.ros/log/9d5167ac-ea4f-11e4-945d-605718ad098a/projectgenerator-3*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [17134]
IDL:omg.org/CORBA/TRANSIENT:1.0
Another rostest instance is currently running, waiting until it finished...
[ROSTEST]test [projectgenerator] finished
[ROSTEST][ROSTEST] test [projectgenerator] done
[ROSTEST]tearDown[/opt/ros/hydro/share/hrpsys_tools/launch/_gen_project.launch]
shutting down processing monitor...
[modelloader-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
... shutting down processing monitor complete
[ROSTEST]rostest teardown /opt/ros/hydro/share/hrpsys_tools/launch/_gen_project.launch complete
ok

----------------------------------------------------------------------
Ran 1 test in 1.226s

OK

WARNING: overall test result is not accurate when --text is enabled
rostest log file is in /home/takasugi/.ros/log/rostest-takasugi-ThinkPad-T450-14544.log
no process to kill
[ 20%] Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JS

Tnoriaki, 2015-04-24:
今日ビルドをしましたが50分経っても終わりませんでした
```
takasugi:hrpsys_ros_bridge_tutorials(master●)$ catkin build --this
-------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/hydro
Workspace:                   /home/takasugi/ros/hydro
Source Space:       [exists] /home/takasugi/ros/hydro/src
Build Space:        [exists] /home/takasugi/ros/hydro/build
Devel Space:        [exists] /home/takasugi/ros/hydro/devel
Install Space:     [missing] /home/takasugi/ros/hydro/install
DESTDIR:                     None
-------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
-------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
-------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------- 
Found '30' packages in 1.1 seconds. 
Starting ==> hrp2_models                                                        
Finished <== hrp2_models                 [ 0.3 seconds ]                        
Starting ==> hrpsys_ros_bridge_tutorials                                        
^C[build] User interrupted, stopping.                                                                                                                                     
                                                                                                                                                                        
[hrpsys_ros_bridge_tutorials] ==> '/home/takasugi/ros/hydro/build/hrpsys_ros_bridge_tutorials/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/takasugi/ros/hydro/build/hrpsys_ros_bridge_tutorials'
Scanning dependencies of target HRP2JSK_WH_SENSORS.urdf_generate
Scanning dependencies of target HRP2JSKNTS_xacro_model_generate
Scanning dependencies of target HRP2JSKNT_WH_SENSORS.urdf_generate
Scanning dependencies of target HRP2JSK_model_generate
[  0%] [  0%] Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.dae
[  0%] [  0%] Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.dae
Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.dae
Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.dae
Humanoid node
Humanoid node
Joint node BASE
Joint node BASE
  Segment node BASE_LINK
  Joint node J1
    Segment node J1_LINK
    Joint node J2
      Segment node J2_LINK
      Joint node J3
        Segment node J3_LINK
        Joint node J4
          Segment node J4_LINK
          Joint node J5
            Segment node J5_LINK
            Joint node J6
              Segment node J6_LINK
              Joint node J7
                Segment node J7_LINK
                Joint node HAND_L
                  Segment node HAND_L_LINK
                Joint node HAND_R
                  Segment node HAND_R_LINK
Humanoid node
Humanoid node
  Segment node BASE_LINK
Joint node BASE
  Joint node J1
[... 3243 lines cut ...]
log file: /home/takasugi/.ros/log/9d5167ac-ea4f-11e4-945d-605718ad098a/projectgenerator-3*.log
[modelloader-2] restarting process
process[modelloader-2]: started with pid [17134]
IDL:omg.org/CORBA/TRANSIENT:1.0
Another rostest instance is currently running, waiting until it finished...
[ROSTEST]test [projectgenerator] finished
[ROSTEST][ROSTEST] test [projectgenerator] done
[ROSTEST]tearDown[/opt/ros/hydro/share/hrpsys_tools/launch/_gen_project.launch]
shutting down processing monitor...
[modelloader-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
... shutting down processing monitor complete
[ROSTEST]rostest teardown /opt/ros/hydro/share/hrpsys_tools/launch/_gen_project.launch complete
ok

----------------------------------------------------------------------
Ran 1 test in 1.226s

OK

WARNING: overall test result is not accurate when --text is enabled
rostest log file is in /home/takasugi/.ros/log/rostest-takasugi-ThinkPad-T450-14544.log
no process to kill
[ 20%] Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HR

Tnoriaki, 2015-04-24:
続きです
```
SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion

NODES
  /
    modelloader (openhrp3/openhrp-model-loader)
[ROSTEST]setup[/opt/ros/hydro/share/hrpsys_tools/launch/_gen_project.launch] run_id[a074f340-ea4f-11e4-a028-605718ad098a] done
[ROSTEST]Running test [projectgenerator]
auto-starting new master
process[master]: started with pid [17738]
ROS_MASTER_URI=http://takasugi-ThinkPad-T450:22422/
setting /run_id to a074f340-ea4f-11e4-a028-605718ad098a
process[rosout-1]: started with pid [17793]
started core service [/rosout]
process[modelloader-2]: started with pid [17809]
[ROSTEST]running test projectgenerator
process[projectgenerator-3]: started with pid [17818]
Warning: Mass is zero. <Model>HRP2JSKNT <Link>RARM_JOINT7
Warning: Mass is zero. <Model>HRP2JSKNT <Link>LARM_JOINT7
Loading body customizer "/opt/ros/hydro/share/OpenHRP-3.1/customizer/libSampleCustomizer.so" for springJoint
hrp::Sensor::FORCE rfsensor lfsensor rhsensor lhsensor
hrp::Sensor::GYRO gyrometer
hrp::Sensor::ACCELERATION gsensor
hrp::Joint RLEG_JOINT0(0) RLEG_JOINT1(1) RLEG_JOINT2(2) RLEG_JOINT3(3) RLEG_JOINT4(4) RLEG_JOINT5(5) RLEG_JOINT6(6) LLEG_JOINT0(7) LLEG_JOINT1(8) LLEG_JOINT2(9) LLEG_JOINT3(10) LLEG_JOINT4(11) LLEG_JOINT5(12) LLEG_JOINT6(13) CHEST_JOINT0(14) CHEST_JOINT1(15) HEAD_JOINT0(16) HEAD_JOINT1(17) RARM_JOINT0(18) RARM_JOINT1(19) RARM_JOINT2(20) RARM_JOINT3(21) RARM_JOINT4(22) RARM_JOINT5(23) RARM_JOINT6(24) RARM_JOINT7(25) LARM_JOINT0(26) LARM_JOINT1(27) LARM_JOINT2(28) LARM_JOINT3(29) LARM_JOINT4(30) LARM_JOINT5(31) LARM_JOINT6(32) LARM_JOINT7(33)
[projectgenerator-3] process has finished cleanly
log file: /home/takasugi/.ros/log/a074f340-ea4f-11e4-a028-605718ad098a/projectgenerator-3*.log
omniNames properly started
Another rostest instance is currently running, waiting until it finished...
[ROSTEST]test [projectgenerator] finished
[ROSTEST][ROSTEST] test [projectgenerator] done
[ROSTEST]tearDown[/opt/ros/hydro/share/hrpsys_tools/launch/_gen_project.launch]
shutting down processing monitor...
[modelloader-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
... shutting down processing monitor complete
[ROSTEST]rostest teardown /opt/ros/hydro/share/hrpsys_tools/launch/_gen_project.launch complete
ok
----------------------------------------------------------------------
Ran 1 test in 1.319s

OK

WARNING: overall test result is not accurate when --text is enabled
rostest log file is in /home/takasugi/.ros/log/rostest-takasugi-ThinkPad-T450-17623.log
[ 22%] Built target hrp2jsknt_hrpsys_ros_bridge_tutorials_compile
[ 22%] Generating /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10_nosim.xml
Starting omniORB omniNames: takasugi-ThinkPad-T450:2951

Fri Apr 24 16:00:45 2015:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000070000000010102000d00000031302e3230382e312e3136360000870b0b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c0000000100000001000100010000000100010509010100010000000901010003545441080000001dea39550100460c
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/takasugi/.ros/log/rostest-takasugi-ThinkPad-T450-17854.log
testprojectgenerator (rostest.runner.RosTest) ... [ROSTEST]setup[/opt/ros/hydro/share/hrpsys_tools/launch/_gen_project.launch] run_id[a18af8ce-ea4f-11e4-870e-605718ad098a] starting
started roslaunch server http://takasugi-ThinkPad-T450:55651/
ros_comm version 1.10.12

SUMMARY
[... 3024 lines cut ...]
[ROSTEST]Running test [projectgenerator]
auto-starting new master
process[master]: started with pid [28144]
ROS_MASTER_URI=http://takasugi-ThinkPad-T450:22422/
setting /run_id to cbc72fd6-ea4f-11e4-a4cc-605718ad098a
process[rosout-1]: started with pid [28203]
started core service [/rosout]
process[modelloader-2]: started with pid [28219]
[ROSTEST]running test projectgenerator
IDL:omg.org/CORBA/TRANSIENT:1.0
omniNames properly started
process[projectgenerator-3]: started with pid [28235]
[modelloader-2] process has finished cleanly
log file: /home/takasugi/.ros/log/cbc72fd6-ea4f-11e4-a4cc-605718ad098a/modelloader-2*.log
respawning...
[modelloader-2] restarting process
process[modelloader-2]: started with pid [28236]
IDL:omg.org/CORBA/TRANSIENT:1.0
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Another rostest instance is currently running, waiting until it finished...
[modelloader-2] process has finished cleanly
log file: /home/takasugi/.ros/log/cbc72fd6-ea4f-11e4-a4cc-605718ad098a/modelloader-2*.log
respawning...
[projectgenerator-3] process has died [pid 28235, exit code -6, cmd /opt/ros/hydro/lib/hrpsys/ProjectGenerator -o corba.nameservers:localhost:2962 /home/takasugi/prog/euslib/rbrain/jaxon/JAXONmain.wrl /opt/ros/hydro/share/OpenHRP-3.1/sample/model/longfloor.w

k-okada, 2015-04-24:
rtmros_common/cmake/compile_robot_mode..
はあたらしいですか．jskeusは1.0.6以上でなければirtrobot.lを更新してください．

--
◉ Kei Okada

On Fri, Apr 24, 2015 at 6:43 PM, Takasugi Noriaki <notifications@github.com>
wrote:

> 今日やり直したら50分経ってもビルドが終わりませんでした
> takasugi:hrpsys_ros_bridge_tutorials(master●)$ catkin build --this
>
> Profile: default
> Extending: [cached] /opt/ros/hydro
> Workspace: /home/takasugi/ros/hydro
> Source Space: [exists] /home/takasugi/ros/hydro/src
> Build Space: [exists] /home/takasugi/ros/hydro/build
> Devel Space: [exists] /home/takasugi/ros/hydro/devel
> Install Space: [missing] /home/takasugi/ros/hydro/install
> DESTDIR: None
>
> Isolate Develspaces: False
> Install Packages: False
> Isolate Installs: False
>
> Additional CMake Args: None
> Additional Make Args: None
> Additional catkin Make Args: None
> Internal Make Job Server: True
>
> Whitelisted Packages: None
> Blacklisted Packages: None Workspace configuration appears valid.
>
> Found '30' packages in 1.1 seconds.
> Starting ==> hrp2_models
>
> Finished <== hrp2_models [ 0.3 seconds ]
>
> Starting ==> hrpsys_ros_bridge_tutorials
>
> ^C[build] User interrupted, stopping.
>
> [hrpsys_ros_bridge_tutorials] ==>
> '/home/takasugi/ros/hydro/build/hrpsys_ros_bridge_tutorials/build_env.sh
> /usr/bin/make --jobserver-fds=3,5 -j' in
> '/home/takasugi/ros/hydro/build/hrpsys_ros_bridge_tutorials'
> Scanning dependencies of target HRP2JSK_WH_SENSORS.urdf_generate
> Scanning dependencies of target HRP2JSKNTS_xacro_model_generate
> Scanning dependencies of target HRP2JSKNT_WH_SENSORS.urdf_generate
> Scanning dependencies of target HRP2JSK_model_generate
> [ 0%] [ 0%] Generating
> /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.dae
> [ 0%] [ 0%] Generating
> /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.dae
> Generating
> /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.dae
> Generating
> /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.dae
> Humanoid node
> Humanoid node
> Joint node BASE
> Joint node BASE
> Segment node BASE_LINK
> Joint node J1
> Segment node J1_LINK
> Joint node J2
> Segment node J2_LINK
> Joint node J3
> Segment node J3_LINK
> Joint node J4
[... 1768 lines cut ...]
> Collada Info: init COLLADA writer version: 1.5.0, namespace:
> http://www.collada.org/2008/03/COLLADASchema
> Collada Info: init COLLADA writer version: 1.5.0, namespace:
> http://www.collada.org/2008/03/COLLADASchema
> Collada Info: joint type free not supported
> Collada Info: joint type free not supported
> /opt/ros/hydro/bin/export-collada
> /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_hrp2/hrp2_models/HRP2W_for_OpenHRP3/HRP2Wmain.wrl
> was successfully exported to
> /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2W.dae
> /opt/ros/hydro/bin/export-collada
> /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_hrp2/hrp2_models/HRP2W_for_OpenHRP3/HRP2Wmain.wrl
> was successfully exported to
> /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2W.dae
> [ 5%] Generating
> /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2jsknts.l
> [ 5%] Generating
> /home/takasugi/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2jsknts.l
> Sensor rfsensor is attached to RLEG_LINK5 base_force6d #sensor1 0
> Sensor lfsensor is attached to LLEG_LINK5 base_force6d #sensor2 1
> Sensor CARMINE is attached to HEAD_LINK1 base_pinhole_camera #sensor5 0
> Sensor rhsensor is attached to RARM_LINK6 base_force6d #sensor6 2
> Sensor lhsensor is attached to LARM_LINK6 base_force6d #sensor7 3
> Sensor gsensor is attached to CHEST_LINK1 base_imu #sensor3 0
> Sensor