jsk-ros-pkg/jsk_model_tools #119: With use-base 0.1, expect PR2 to use base and move around.

aginika, 2015-06-22:
I expect ":use-base" will make PR2 use kind of move-base to solve the Inverse-kinematics.
But at the below code, PR2 just move the arm as "8" and pr2's base doesn't move.(Torso is working)
Even if I move the target coords far from PR2, this just ended to fail Inverse-kinematics. It didn't move.

https://github.com/jsk-ros-pkg/jsk_model_tools/blob/065556181e1ac10cd2103e9dc7ca3cad7f83e0de/euscollada/test/euscollada-pr2-test.l#L25

(I also needed to add ":move-target" ":link-list" to prevent some errors)

k-okada, 2015-06-22:
please look into pr2-util.l and jskeus

aginika, 2015-06-22:
Thank you!
With pr2util.l, it use the base of PR2, but the many of Inverse Kinematics failed...

I test some of parameters
Not Use-base
[image]

Use-base 0.1
[image]

Use-base 0.5
[image]

Use-base 0.9
[image]

Use-base 0.99
[image]

Use-base 10 was as same as 0.99

What can I do with these things?... It seems to just add the base to addtional-weight-list.
Do I need to move the pr2 as near to the target as possible by myself?

k-okada, 2015-06-22:
what is expected behavior? all box should be red?
please run ik with :debug t and see what's going on. Also commit your code
under pr2 as test program and create PR

aginika, 2015-06-22:
Yes, I hope all the box become red.
Ok, I will.

closed 2015-06-22 by aginika

k-okada, 2015-06-22:
If would debug as follows:

1 attach end coords directory to the base, and solve IK, this is 3dof(x,y,yaw) robot should find solution
2 use arm end coords but set small weight ( I suppose use greater number for eus arguments?) not to move arms . Of cause, arm initial arm postures is set to grab object no the ground. From the kinematics point of view, this is almost equals to 1. So it should move.
3 if 2 works, your program should work, with small modification.

Good luck!
--
◉ Kei Okada

aginika, 2015-06-23:
Thank you. I will.

As a reference this is the behavior of the IK with use-base 0.99,
[image]

h-kamada, 2015-06-23:
I was also using :use-base and encountered the same problem.
And, I was searching for the file in which the :inverse-kinematics method is written and adding weight is defined.
I couldn't find where something about adding weight is defined.

super class of pr2-robot is euscollada-robot and it's written in euscollada/src/euscollada-robot.l.
super class of euscollada-robot is robot-model and it's written in euslisp/jskeus/irteus/irtrobot.l.
super class of robot-model is cascaded-link and it's written in euslisp/jskeus/irteus/irtmodel.l.

I thought something about adding weight is written in above files. But, I couldn't find it.
Where is it defined?

snozawa, 2015-06-23:
Did you check `pr2eus/pr2-utils.l`?

h-kamada, 2015-06-23:
yes. But, in that file, detail of that is not written.
Word additional-weight-list is in irtmodel or irtrobot, so detail may be in that file.
But, structure of function is complex, so couldn't find.

snozawa, 2015-06-23:
I see.
Learning from `irtmodel.l` will help you.
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l

In inverse-kinematics calculation, `:calc-inverse-kinematics-weight-from-link-list` is called:
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L1716
Then, calculated weight value is used in SR-inverse calculation:
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L1045

`:calc-inverse-kinematics-weight-from-link-list` returns weighting vector:
https://github.com/euslisp/jskeus/blob/master/irteus/irtmodel.l#L1548
`additional-weight-list` directly changes weighting value for specified joints in this method.

If you want to weight vector value in each iteration, please use `:debug-view t`.

h-kamada, 2015-06-23:
Thank you very much!