jsk-ros-pkg/jsk_model_tools #123: オブジェクトに応じてgazeboのstatic/fixed/無印属性を変更したworldファイルを生成したい

furushchev, 2015-07-09:
https://github.com/jsk-ros-pkg/jsk_model_tools/pull/122
で
https://github.com/euslisp/EusLisp/tree/master/models
のモデルが自動でurdfに変換されるようになりましたが、作成されたworldファイルを見ると
https://gist.github.com/94c2253a1b1f81266a11
すべてのmodelファイルに`_static`とついています。
modelsフォルダには
- room610-tv3-0
- room610-tv3-0_fixed
- room610-tv3-0_static

のように3つのurdfができていて、それぞれ無印に比べて

- `_fixed`:
  
```diff
+  <!-- Used for fixing robot to Gazebo 'base_link' -->
+  <link name="world"/>
+  <joint name="fixed" type="fixed">
+    <parent link="world"/>
+    <child link="nil_link"/>
+  </joint>
```

- `_static`:

```diff
-    <static>false</static>
+    <static>true</static>
```

というようになっています。

`static`はworldの指定位置に固定、`fixed`はworldに固定するけど、例えばタンスの引き出しなどのリンクは可動にするための処置と見られます。

今回eusモデルから変換すると一括ですべてstaticになっているところを、キッチンなどはstatic, 棚などはfixed, トレイなどは無印というように変えたいのですが、そのような指定をどこに書けばいいでしょうか？

mmurooka, 2015-07-09:
eusurdfに設定ファイルをおいて
https://github.com/jsk-ros-pkg/jsk_model_tools/blob/master/eusurdf/euslisp/convert-eus-to-urdf.l#L69
で_staticとあるところを，その設定ファイルに応じて変えるのがいいと思います．
モデルが飛んでいってしまうので，デフォルトはstaticがお勧めです．

furushchev, 2015-07-09:
@mmurooka なるほど。eusurdfに設定ファイルをおくということはeuslisp/modelsのモデルファイル一覧に対してホワイトリストかブラックリストを作るという事でしょうか？
その場合、目的は実現可能ですが、その設定ファイルをまたeuslisp/modelsから自動生成したくなってくる気がします。。。

mmurooka, 2015-07-09:
eusの方針はちゃんと分かっていないですが，
大変そうですがそれでもいいのだと思います．

> eusモデルに質量モデルと位置をきちんと設定して

床や壁は質量・位置が正しくても動いてほしくないので，
eusモデルの質量・位置が正しくてもどこか（eusモデルか変換時か）でstaticにするという情報を加える必要はありそうです．