jsk-ros-pkg/jsk_model_tools #124: eusurdfのモデル変換で/tmpにファイルを作るが他のユーザが上書きできない

mmurooka, 2015-07-15:
特にPR2で起きていますが，
eusからurdfにモデル変換するときに/tmpに途中生成のcolladaファイルができるようになっていて，
ユーザaaaが変換したcolladaが/tmpに残っていると，
ユーザbbbが同じ名前のcolladaファイルを/tmpに作れず，止まってしまう，
という問題がおきています．

どのように解決するのがよろしいでしょうか．
１．コンパイルのたびに，/tmpにランダムな名前のディレクトリを作って，そこにcolladaを置く
２．colladaファイルを作るときに誰でも消せるように権限を付ける
くらいしか思いつかず
１．でやってみようかと思いますがよろしいでしょうか．

garaemon, 2015-07-15:
/tmpのしたにmktempでもう一回層ディレクトリを掘るんじゃないかな?

mmurooka, 2015-07-15:
ありがとうございます．

eusでランダムに作ったディレクトリの名前を知りたいのですが，
良い方法ありますでしょうか．
eusのプログラムの中でディレクトリを作るので構いません．

snozawa, 2015-07-15:
> eusでランダムに作ったディレクトリの名前を知りたいのですが，
> 良い方法ありますでしょうか．
> eusのプログラムの中でディレクトリを作るので構いません．

これが必要な状況は、ちょっと複雑な状況かな、とおもいます。
変換のフロー図をみなおしてみて、もうちょい別な解法があるかも検討してみるとよいですね。
なんとか途中生成ファイルをお互いにやりとりしなくてすむように、などなど

それで、上記をもしやるとしたらなにかコンフリクトしない名前づけでルール決めとくとかはあるかもしれません。
eusのプロセスのPIDの名前をつける、とか。

garaemon, 2015-07-15:
乱数で適当に数字をくっつけて、そのファイルが被っていないかを確認して繰り返し行う、とか色々とやり方はある気がします。

mmurooka, 2015-10-27:
https://github.com/jsk-ros-pkg/jsk_model_tools/pull/126 で解決しているのでcloseします．

closed 2015-10-27 by mmurooka