jsk-ros-pkg/jsk_model_tools #140: Request for new version release

snozawa, 2015-08-31:
Request for new version release to update euscollada-robot.l for downsteram tests.

garaemon, 2015-08-31:
I will

garaemon, 2015-08-31:
0.1.13 https://github.com/jsk-ros-pkg/jsk_model_tools/commit/fb77908f35e49292d2e8028432bf0ddb1aafe7ed

garaemon, 2015-08-31:
https://mail.google.com/mail/u/0/#inbox/14f83d127a03e9f4
released

deb will be released in one or two days.

closed 2015-08-31 by garaemon

k-okada, 2015-08-31:
> 0.1.13 fb77908

ran bloom -> https://github.com/ros/rosdistro/pull/9339

k-okada, 2015-08-31:
@snozawa

you can use
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/add_tag/build?delay=0sec to
release by yourself
select repository and push build button, and check with console output (
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/add_tag/131/console), if
that is ok, chosse "FALSE" and build again