jsk-ros-pkg/jsk_model_tools #143: [travis] Compile jsk_model_tools on jade

wkentaro, 2015-09-14:
travis test on jade does not pass with below error:

```
[eus_assimp] --   found SDL_image, version 1.2.12                               

[eusurdf] -- Using these message generators: gencpp;geneus;genlisp;genpy        

[euscollada] -- Using these message generators: gencpp;geneus;genlisp;genpy     

[euscollada] CMake Error at /opt/ros/jade/share/catkin/cmake/catkinConfig.cmake:75 (find_package): 

[euscollada]   Could not find a package configuration file provided by "urdfdom" with any 

[euscollada]   of the following names:                                          

[euscollada]                                                                    

[euscollada]     urdfdomConfig.cmake                                            

[euscollada]     urdfdom-config.cmake                                           

[euscollada]                                                                    

[euscollada]   Add the installation prefix of "urdfdom" to CMAKE_PREFIX_PATH or set 

[euscollada]   "urdfdom_DIR" to a directory containing one of the above files.  If 

[euscollada]   "urdfdom" provides a separate development package or SDK, be sure it has 

[euscollada]   been installed.                                                  

[euscollada] Call Stack (most recent call first):                               

[euscollada]   CMakeLists.txt:8 (find_package)                                  

[euscollada]                                                                    

[euscollada]                                                                    

[euscollada] -- Configuring incomplete, errors occurred!                        

[euscollada] See also "/workspace/ros/ws_jsk_model_tools/build/euscollada/CMakeFiles/CMakeOutput.log". 

[euscollada] See also "/workspace/ros/ws_jsk_model_tools/build/euscollada/CMakeFiles/CMakeError.log". 
```

My ros package depends on this package and has problem.
https://github.com/start-jsk/jsk_picking_challenge/pull/836

wkentaro, 2015-09-14:
The working PR is here: 117

garaemon, 2015-09-14:
in jade, urdfdom is not available

```
$ ROS_DISTRO=jade rosdep db | grep urdf  
liburdfdom-tools -> liburdfdom-tools
```

wkentaro, 2015-09-14:
fwd: @k-okada

wkentaro, 2015-09-14:
the difference seems

* eusurdf -> ros-indigo-eusurdf
* iai_urdf_msgs -> ros-indigo-iai-urdf-msgs

```
% rosdep db | grep urdf | sort
collada_urdf_jsk_patch -> ros-indigo-collada-urdf-jsk-patch
collada_urdf -> ros-indigo-collada-urdf
eusurdf -> ros-indigo-eusurdf
iai_urdf_msgs -> ros-indigo-iai-urdf-msgs
liburdfdom-dev -> liburdfdom-dev
liburdfdom-headers-dev -> liburdfdom-headers-dev
liburdfdom-tools -> liburdfdom-tools
p2os_urdf -> ros-indigo-p2os-urdf
urdfdom_py -> ros-indigo-urdfdom-py
urdf_parser_plugin -> ros-indigo-urdf-parser-plugin
urdf -> ros-indigo-urdf
urdf_tutorial -> ros-indigo-urdf-tutorial

% ROS_DISTRO=jade rosdep db | grep urdf | sort
collada_urdf_jsk_patch -> ros-jade-collada-urdf-jsk-patch
collada_urdf -> ros-jade-collada-urdf
liburdfdom-dev -> liburdfdom-dev
liburdfdom-headers-dev -> liburdfdom-headers-dev
liburdfdom-tools -> liburdfdom-tools
p2os_urdf -> ros-jade-p2os-urdf
urdfdom_py -> ros-jade-urdfdom-py
urdf_parser_plugin -> ros-jade-urdf-parser-plugin
urdf -> ros-jade-urdf
urdf_tutorial -> ros-jade-urdf-tutorial
```

wkentaro, 2015-09-14:
In first place, I cannot find urdfdomConfig.cmake in my env.
```
% rospack list | grep urdf
collada_urdf /opt/ros/indigo/share/collada_urdf
collada_urdf_jsk_patch /opt/ros/indigo/share/collada_urdf_jsk_patch
urdf /opt/ros/indigo/share/urdf
urdf_parser_plugin /opt/ros/indigo/share/urdf_parser_plugin
urdf_tutorial /opt/ros/indigo/share/urdf_tutorial
urdfdom_py /opt/ros/indigo/share/urdfdom_py
```

wkentaro, 2015-09-15:
The problem seems because euscollada does not consider about jade.
there is no condition after indigo.
https://github.com/jsk-ros-pkg/jsk_model_tools/blob/master/euscollada/CMakeLists.txt#L5-L9

commit f59ada56 (2015-09-17): Add travis test for ROS jade

Closes #143
closed 2015-09-17 by garaemon