jsk-ros-pkg/jsk_model_tools #154: eus2colladaの使い方

YuOhara, 2015-11-11:
```
1.irteusgl$ load "package://euscollada/src/eus2collada.l"
t
2.irteusgl$ eus2collada "tray-object.l" "/home/leus"
```

のようにすると、

https://github.com/jsk-ros-pkg/jsk_model_tools/blob/master/euscollada/src/eus2collada.l#L353
のところで
```
/home/leus/ros/indigo_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusg\
l 0 error: unbound variable articulated-link in (derivedp obj articulated-link)
```

とエラーになってしまうのですが、どうしたらよいでしょうか？

* 使い方が間違っている
* 昔あったクラスがない

YuOhara, 2015-11-11:
クラス自体は、grepすると
https://github.com/euslisp/jskeus/blob/master/irteus/demo/demo.l#L29
に同じ名前のものがありました。

mmurooka, 2015-11-11:
ファイル名ではなくオブジェクトのインスタンスを渡すのではないでしょうか．

YuOhara, 2015-11-11:
確かにそうでしたね。
しかしやはり、クラスはないとエラーが出ます。
```
6.E2-irteusgl$ (setq a (instance tray-object :init))
#<tray-object #X677c8c8 tray  0.0 0.0 0.0 / 0.0 0.0 0.0>
7.E2-irteusgl$ objects a
;; (make-irtviewer) executed
(#<tray-object #X677c8c8 tray  0.0 0.0 0.0 / 0.0 0.0 0.0>)
8.E2-irteusgl$ eus2collada a "/home/leus"
/home/leus/ros/indigo_parent/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: unbound variable articulated-link in (derivedp obj articulated-link)
```

k-okada, 2015-11-11:
articulated-linkはeuslibの関数でirteusにはありません．
まずはirtuesにコンバートしたらいいです．最近は @furschev (スペル直して）がプロです

mmurooka, 2015-11-12:
@YuOhara の本来の目的はeusのオブジェクトをurdfにすることだそうなので，
euslibでモデルを作ってirteusの変換した後は，

https://github.com/mmurooka/jsk_model_tools/blob/fix-convert-eus-to-urdf/eusurdf/README.md#convert-eus-model-to-urdf
のようにすると，irteusのモデルをurdfに変換できます．
（中でeus2colladaを呼んでいます．）

YuOhara, 2016-02-03:
ありがとうございます。上で書かれている方法で、urdfに変換できました。

closed 2016-02-03 by YuOhara