jsk-ros-pkg/jsk_model_tools #162: Failed to release new version

garaemon, 2015-12-08:
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/bloom_release/5874/console
```
15:44:22  * [new tag]         debian/ros-jade-eus-assimp_0.2.1-4_trusty -> debian/ros-jade-eus-assimp_0.2.1-4_trusty
15:44:22  * [new tag]         debian/ros-jade-eus-assimp_0.2.1-4_utopic -> debian/ros-jade-eus-assimp_0.2.1-4_utopic
15:44:22  * [new tag]         debian/ros-jade-eus-assimp_0.2.1-4_vivid -> debian/ros-jade-eus-assimp_0.2.1-4_vivid
15:44:22  * [new tag]         debian/ros-jade-euscollada_0.2.1-4_trusty -> debian/ros-jade-euscollada_0.2.1-4_trusty
15:44:22  * [new tag]         debian/ros-jade-euscollada_0.2.1-4_utopic -> debian/ros-jade-euscollada_0.2.1-4_utopic
15:44:22  * [new tag]         debian/ros-jade-euscollada_0.2.1-4_vivid -> debian/ros-jade-euscollada_0.2.1-4_vivid
15:44:22  * [new tag]         debian/ros-jade-eusurdf_0.2.1-4_trusty -> debian/ros-jade-eusurdf_0.2.1-4_trusty
15:44:22  * [new tag]         debian/ros-jade-eusurdf_0.2.1-4_utopic -> debian/ros-jade-eusurdf_0.2.1-4_utopic
15:44:22  * [new tag]         debian/ros-jade-eusurdf_0.2.1-4_vivid -> debian/ros-jade-eusurdf_0.2.1-4_vivid
15:44:22  * [new tag]         debian/ros-jade-jsk-model-tools_0.2.1-4_trusty -> debian/ros-jade-jsk-model-tools_0.2.1-4_trusty
15:44:22  * [new tag]         debian/ros-jade-jsk-model-tools_0.2.1-4_utopic -> debian/ros-jade-jsk-model-tools_0.2.1-4_utopic
15:44:22  * [new tag]         debian/ros-jade-jsk-model-tools_0.2.1-4_vivid -> debian/ros-jade-jsk-model-tools_0.2.1-4_vivid
15:44:22  * [new tag]         release/jade/eus_assimp/0.2.1-4 -> release/jade/eus_assimp/0.2.1-4
15:44:22  * [new tag]         release/jade/euscollada/0.2.1-4 -> release/jade/euscollada/0.2.1-4
15:44:22  * [new tag]         release/jade/eusurdf/0.2.1-4 -> release/jade/eusurdf/0.2.1-4
15:44:22  * [new tag]         release/jade/jsk_model_tools/0.2.1-4 -> release/jade/jsk_model_tools/0.2.1-4
15:44:22  * [new tag]         rpm/ros-jade-eus-assimp-0.2.1-4_21 -> rpm/ros-jade-eus-assimp-0.2.1-4_21
15:44:22  * [new tag]         rpm/ros-jade-eus-assimp-0.2.1-4_22 -> rpm/ros-jade-eus-assimp-0.2.1-4_22
15:44:22  * [new tag]         rpm/ros-jade-euscollada-0.2.1-4_21 -> rpm/ros-jade-euscollada-0.2.1-4_21
15:44:22  * [new tag]         rpm/ros-jade-euscollada-0.2.1-4_22 -> rpm/ros-jade-euscollada-0.2.1-4_22
15:44:22  * [new tag]         rpm/ros-jade-eusurdf-0.2.1-4_21 -> rpm/ros-jade-eusurdf-0.2.1-4_21
15:44:22  * [new tag]         rpm/ros-jade-eusurdf-0.2.1-4_22 -> rpm/ros-jade-eusurdf-0.2.1-4_22
15:44:22  * [new tag]         rpm/ros-jade-jsk-model-tools-0.2.1-4_21 -> rpm/ros-jade-jsk-model-tools-0.2.1-4_21
15:44:22  * [new tag]         rpm/ros-jade-jsk-model-tools-0.2.1-4_22 -> rpm/ros-jade-jsk-model-tools-0.2.1-4_22
15:44:23 ROS Distro index file associate with commit '7cb1b8959ddbb7de4137857604e901912673d6a7'[0m
15:44:23 New ROS Distro index url: 'https://raw.githubusercontent.com/ros/rosdistro/7cb1b8959ddbb7de4137857604e901912673d6a7/index.yaml'[0m
15:44:23 [32m[1m==> [0m[0mFetching 'jsk_model_tools' repository from 'https://github.com/tork-a/jsk_model_tools-release.git'[0m
15:44:23 [32m[1m==> [0mTesting for push permission on release repository[0m[0m
15:44:23 [34m[1m==> [0m[1m[0mgit remote -v[0m
15:44:23 [34m[1m==> [0m[1m[0mgit push --dry-run[0m
15:44:23 [32m[1m==> [0m[0mReleasing 'jsk_model_tools' using release track 'jade'[0m
15:44:23 [34m[1m==> [0m[1mgit-bloom-release jade[0m[0m
15:44:23 [32m<== [0m[0mReleased 'jsk_model_tools' using release track 'jade' successfully[0m
15:44:23 [32m[1m==> [0m[0mPushing changes to release repository for 'jsk_model_tools'[0m
15:44:23 [34m[1m==> [0m[1mgit push --all[0m[0m
15:44:23 [32m<== [0m[0mPushed changes successfully[0m
15:44:23 [32m[1m==> [0m[0mPushing tags to release repository for 'jsk_model_tools'[0m
15:44:23 [34m[1m==> [0m[1mgit push --tags[0m[0m
15:44:23 [32m<== [0m[0mPushed tags successfully[0m
15:44:23 [32m[1m==> [0m[0mGenerating pull request to distro file located at 'https://raw.githubusercontent.com/ros/rosdistro/7cb1b8959ddbb7de4137857604e901912673d6a7/jade/distribution.yaml'[0m
15:44:23 [31m[1mFailed to open pull request: error - (25, 'Inappropriate ioctl for device')[0m
```

k-okada, 2015-12-12:
released new version and now succeeded to build deb file  http://jenkins.ros.org/job/ros-indigo-eusurdf_sourcedeb/

garaemon, 2015-12-14:
Looks still failed on jsk jenkins.

garaemon, 2015-12-14:
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/bloom_release/5876/console