jsk-ros-pkg/jsk_model_tools #171: objects (send *pr2* :larm :links) に全ての腕のリンクが入らない

k-okada, 2016-02-04:
`(objects (send *pr2* :larm :links)) `しても全ての腕のリンクが表示されず以下のようになります．

理由はlinks には non fixed joint の car が入っていて，そこからassoc されているlinkは入っていないんですが，これにより，
https://github.com/jsk-ros-pkg/jsk_pr2eus/issues/206
の問題が起きています．

うまい解決法はないでしょうか．あるいはもう解決されていますでしょうか？　@YoheiKakiuchi , @snozawa 

[image]

```
45.irteusgl$ objects (send *pr2* :larm :links)
(#<bodyset-link #X7d5dbb8 l_shoulder_pan_link  -50.0 188.0 922.458 / -0.487 0.0 0.0> #<bodyset-link #X68314f0 l_shoulder_\
lift_link  38.362 141.179 922.458 / -0.487 1.059 5.551e-17> #<bodyset-link #X6831a18 l_upper_arm_roll_link  38.362 141.17\
9 922.458 / -0.487 1.059 -0.502> #<bodyset-link #X67d2788 l_elbow_flex_link  211.396 49.492 573.671 / -0.098 0.232 -0.244\
> #<bodyset-link #X6658718 l_forearm_roll_link  211.396 49.492 573.671 / -0.098 0.232 0.266> #<bodyset-link #X7d01a08 l_w\
rist_flex_link  522.317 18.806 499.998 / -0.161 8.432e-06 0.258> #<bodyset-link #X76bde28 l_wrist_roll_link  522.317 18.8\
06 499.998 / -0.161 8.432e-06 -3.036e-06>)
46.irteusgl$ 
```

YoheiKakiuchi, 2016-02-05:
対応できていません。
```send *robot* limb :links :joint-list```
として出てるくるリンクやジョイントはrobot.yamlに
書かれているものだけになっています。

これについては、
euscollada-robotの定義の以下のcollision modelを作っている部分で
https://github.com/jsk-ros-pkg/jsk_model_tools/blob/master/euscollada/src/euscollada-robot.l#L64-L95
fixed jointでつながっているリンクも元リンクにつなげてしまえばいいのではないかと思います。
問題としては、collisionの形状とobjectsで表示される形状が違ってきてしまうことがありそうです。

同じような問題ですが、ハンドのリンクも出ないです。
```(send *pr2* :gripper :larm :links)```で返ってきても良さそうですが、
メソッドではlimbのリンクの最後にend-coordsがassocされていることが前提で、
PR2ではハンドの先にend-coordsがassocされています。
これは、以下をarmの最終リンクに修正すれば解決するように思います。
https://github.com/jsk-ros-pkg/jsk_model_tools/blob/master/euscollada/pr2.yaml#L31

k-okada, 2016-04-27:
cc: @h-kamada

k-okada, 2020-05-26:
https://gist.github.com/k-okada/2d0e4eacd2f0f5278fa7 にサンプルプログラムがある（がドキュメント不足．．．）