jsk-ros-pkg/jsk_model_tools #228: When I launch gazebo-73b2 urdf.xacro model, Gazebo crashed

mqcmd196, 2020-05-15:
I executed
```bash
roslaunch gazebo_ros empty_world.launch
roslaunch eusurdf gazebo_spawn_scene.launch scene:=room73b2
```
as the readme shows. I applied the force on the door, Gazebo crashed.

When I used simple model (https://github.com/jsk-ros-pkg/jsk_demos/pull/1295/commits/77b19b905b131b72ca8ccedaa6bfc27c6bd25738), it didn't crash.

Thanks.

knorth55, 2020-05-15:
@mqcmd196 
we need more information for debugging.
can you paste error log and also gif image of your case?
you can use kazam for recording your screen.

mqcmd196, 2020-05-15:
I didn't know how to convert mp4 to gif, I uploaded on gdrive.
https://drive.google.com/file/d/1z_RQlyG6FK30zCVGDq5cqJz_JsFXAwA5/view

0:00 ~ 0:15 Launch gazebo and xacro world
0:28 Apply torque to the door (It was succeeded at first)
0:43 After I choose the wall, I couldn't choose the door and it crashed. I waited for few minutes, but there are no responds from Gazebo.