jsk-ros-pkg/jsk_model_tools #234: [eusurdf] cascaded-linkと，それにassocするbodyset-linkの原点の位置がずれていると上手く変換されない

ketaro-m, 2021-05-10:
cascaded-linkにassocされているbodyset-linkの原点がcascaded-linkの原点とずれていると，生成されたurdf上でそのbodyset-linkの原点がcascaded-linkの原点に移動させられ，その影響が伝播してその先のbody-linkらの原点も順にずれてしまいます．

エラーの出たソースコードは[ここ](https://gist.github.com/ketaro-m/bfb5e0ddc6971166929417f06be7039b)にあり，wheelchair_error.l をconvert-eus-to-urdf.lで変換すると，以下の写真のように軸の位置がずれてしまいます．
[image]

wheelchair_fixed.lを変換すると，正しい軸の位置のまま変換されます．
[image]

wheelchair_error.l と wheelchair_fixed.l のdiffは[この部分](https://gist.github.com/ketaro-m/bfb5e0ddc6971166929417f06be7039b#file-wheelchair_error-l-L257-L258)で，wheelchair_error.lの方ではcascaded-linkにassocしているbodyset-linkの原点を移動させているのですが，変換後のurdfではこの原点が強制的にcascaded-linkの原点と同じ位置に戻されてしまっているため，その先の子リンクたちの原点も順次移動してしまっているのではないかと考えられます．
wheelchair_fixed.l の方ではbodyset-linkの原点はそのままcascaded-linkの原点と一致したままでbodyの位置のみを移動させているために，このようなエラーは出ていません．

上記のようにcascaded-linkとそこにassocするbodyset-linkの原点がずれている場合に加え，cascaded-linkに複数のbodyset-linkをassocした場合も何かしらのエラーが出るのではないかと考えられます．

YoheiKakiuchi, 2021-05-16:
urdfへ変換をするのに、以下のように行われています。
```eus-model -(irtcollada.l)-> COLLADA形式 -(collada_to_urdf)-> URDF形式```

それで今回は、irtcollada.lの段階でバグがあったようで、
このPRで修正したつもりです https://github.com/euslisp/jskeus/pull/600

あと、この問題と直接は関係ないのですが、wheelchair.lのモデルで、
jointのインスタンスをバインドしている変数がグローバル変数になっています。
これらの値 ```jlleg jltire jlstire jrleg jrtire jrstire```

letで宣言してローカル変数にするか、wheelchairクラスの ```:slots```に宣言して
インスタンス変数にするのが良いかと思います。

ファイルをコンパイルするとグローバル変数があった場合warningが出るので確認できます。
```(comp::compile-file "wheelchair.l")```

ketaro-m, 2021-05-17:
ありがとうございます．マージされたら試させていただきます．

jointのインスタンスはスロット変数にして，コンパイルのwarningも出なくなりました
ご指摘ありがとうございます．