jsk-ros-pkg/jsk_model_tools #248: [euscollada] noeticでeuscolladaをbuildするとurdfdomを見つけられずに失敗する

tkmtnt7000, 2022-07-25:
noeticでeuscolladaをビルドしようとすると以下のようにエラーが出て失敗します。

Reproducing
```
docker run -it -rm --name hoge ros:noetic 
apt update
apt install sudo git # for docker
rosdep update
mkdir ros/robot_ws/src -p
cd ros/robot_ws/src
git clone https://github.com/jsk-ros-pkg/jsk_model_tools
rosdep install -y -r --from-paths . --ignore-src
cd ../
source /opt/ros/noetci/setup.bash
catkin init
catkin build euscollada
```
log
```
root@d533dce0e041:/ros/robot_ws# catkin b euscollada
==> Expanding alias 'b' from 'catkin b euscollada' to 'catkin build euscollada'
--------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/noetic
Workspace:                   /ros/robot_ws
--------------------------------------------------
Build Space:        [exists] /ros/robot_ws/build
Devel Space:        [exists] /ros/robot_ws/devel
Install Space:      [unused] /ros/robot_ws/install
Log Space:         [missing] /ros/robot_ws/logs
Source Space:       [exists] /ros/robot_ws/src
DESTDIR:            [unused] None
--------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
--------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------
[build] Found 4 packages in 0.0 seconds.                                                                                               
[build] Updating package table.                                                                                                        
Starting  >>> catkin_tools_prebuild                                                                                                    
Finished  <<< catkin_tools_prebuild                [ 1.1 seconds ]                                                                     
Starting  >>> euscollada                                                                                                               
_______________________________________________________________________________________________________________________________________
Errors     << euscollada:cmake /ros/robot_ws/logs/euscollada/build.cmake.000.log                                                       
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "urdfdom" with any
  of the following names:

    urdfdomConfig.cmake
    urdfdom-config.cmake

  Add the installation prefix of "urdfdom" to CMAKE_PREFIX_PATH or set
  "urdfdom_DIR" to a directory containing one of the above files.  If
  "urdfdom" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:22 (find_package)

cd /ros/robot_ws/build/euscollada; catkin build --get-env euscollada | catkin env -si  /usr/bin/cmake /ros/robot_ws/src/jsk_model_tools/euscollada --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/ros/robot_ws/devel/.private/euscollada -DCMAKE_INSTALL_PREFIX=/ros/robot_ws/install; cd -

.......................................................................................................................................
Failed     << euscollada:cmake                     [ Exited with code 1 ]                                                              
Failed    <<< euscollada                           [ 1.0 seconds ]                                                                     
[build] Summary: 1 of 2 packages succeeded.                                                                                            
[build]   Ignored:   3 packages were skipped or are skiplisted.                                                                        
[build]   Warnings:  None.                                                                                                             
[build]   Abandoned: None.                                                                                                             
[build]   Failed:    1 packages failed.                                                                                                
[build] Runtime: 2.1 seconds total.                                                                                                    
[build] Note: Workspace packages have changed, please re-source setup files to use them

oot@d533dce0e041:/ros/robot_ws# apt search urdfdom
Sorting... Done
Full Text Search... Done
liburdfdom-dev/focal,now 1.0.4+ds-2 amd64 [installed]
  URDF DOM - development files

liburdfdom-headers-dev/focal,now 1.0.4-1 amd64 [installed,automatic]
  URDF DOM - header files

liburdfdom-model/focal,now 1.0.4+ds-2 amd64 [installed,automatic]
  URDF DOM - model library

liburdfdom-model-state/focal,now 1.0.4+ds-2 amd64 [installed,automatic]
  URDF DOM - model state library

liburdfdom-sensor/focal,now 1.0.4+ds-2 amd64 [installed,automatic]
  URDF DOM - sensor library

liburdfdom-tools/focal 1.0.4+ds-2 amd64
  URDF DOM - tools

liburdfdom-world/focal,now 1.0.4+ds-2 amd64 [installed,automatic]
  URDF DOM - world library

ros-noetic-urdfdom-py/focal 0.4.6-1focal.20220107.000752 amd64
  Python implementation of the URDF parser.

```

一時的な解決法として
https://github.com/jsk-ros-pkg/jsk_model_tools/blob/6ecd7244f0e1ee0d25e4f458226d2d4b64285a37/euscollada/CMakeLists.txt#L22
でurdfdomを削除してビルドするとビルドは通る。

708yamaguchi, 2022-07-25:
実行しただけですが、僕のUbuntu 18.04のPCでも同じエラーが出ますね
```
$ uname -a
Linux b4apc 5.4.0-120-generic #136~18.04.1-Ubuntu SMP Fri Jun 10 18:00:44 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

細かいタイポの指摘ですが、
```
docker run -it -rm --name hoge ros:noetic 
```
の-rmは--rmで、
```
source /opt/ros/noetci/setup.bash
```
は
noetciがnoetic
でしょうか。

tkmtnt7000, 2022-07-25:
すみません。タイポしていたので編集しました。

k-okada, 2022-07-25:
https://github.com/jsk-ros-pkg/jsk_model_tools/pull/247
をマージしたのでもう一度試してください．ただ，時間は有限なので，この作業の優先度はぜひ考えてみましょう．